KFERMercer/OpenWrt-CI #9: 最新代码出来的固件反而变得更旧了。。。

adaaaaaa, 2019-12-04:
老哥，用你的最新代码之后，更新了代码之后，编译的固件，反而更老了。。。
什么原因啊？昨天还正常的，今天出来变成R11.23最早的版本了。。。

KFERMercer, 2019-12-04:
因为你手动改过`package/lean/default-settings/files/zzz-default-settings`里的`DISTRIB_REVISION`.

合并机制会人为这是"你的"修改而略过合并此项, 因而版本号不会变.

这涉及到git merge机制问题. 这个问题不需要也没办法解决, 除非你知道怎样手动合并.

对于小白来说, 不要修改非必要文件, 除非你知道自己在做什么.

closed 2019-12-04 by KFERMercer

KFERMercer, 2019-12-04:
看了下你的代码, 实际上是因为你的openwrt-ci是用的别人的修改版. 既然你已经把自动更新CI给删了, 就不用管这么多了.

adaaaaaa, 2019-12-06:
> 我愿意为问题的提出者 @adaaaaaa 定制一个云编译CI.

可以换成教我怎么弄个docker里的openwrt吗？L大，源码的都行。。。😂😂🤪

adaaaaaa, 2019-12-06:
好像docker镜像仓库里面并没有L大的镜像。。。

KFERMercer, 2019-12-06:
官方文档:
https://openwrt.org/start?id=zh/docs/guide-user/virtualization/docker_openwrt_image

adaaaaaa, 2019-12-06:
> 官方文档

在openwrt的docker里运行linux好，还是反过来好？？😂😂😂

adaaaaaa, 2019-12-06:
https://www.mlapp.cn/376.html这里提到：
由于容器内系统内核共享宿主机内核，所以在 Luci 面板的 “内核版本”项与宿主机内核版本一致；
linux主机的内核升级会导致op崩溃吗？或者变得不好使？反过来呢？

KFERMercer, 2019-12-06:
> 在openwrt的docker里运行linux好，还是反过来好？？😂😂😂

反过来. 不光如此, 我甚至严重怀疑在docker中运行op的意义所在.

> https://www.mlapp.cn/376.html这里提到：
> 由于容器内系统内核共享宿主机内核，所以在 Luci 面板的 “内核版本”项与宿主机内核版本一致；
> linux主机的内核升级会导致op崩溃吗？或者变得不好使？反过来呢？

不会. 不会. 不会.

adaaaaaa, 2019-12-08:
在docker里装op能实现所有的功能吗？为什么你觉得docker中运行op意义不大呢？不是性能损失更少吗？

KFERMercer, 2019-12-08:
简单告诉你, 意义不大.