KFERMercer/OpenWrt-CI #11: 关于定制编译问题

fzsimlu, 2019-12-05:
你好，我想问下如果我是定制硬路由的固件，比如newifi-d2 ,我应该怎么调整代码，具体是命名规制之类的是怎么样，有什么参考资料吗。

【正确的定制固件脚本应该是长这样的, 此配置基于大雕最新代码, 开启了所有预先写好的定制选项, 同时已经经过测试可用. 悟性再高不如照葫芦画瓢, 如果你的配置一直编译失败, 抄, 请.】

我尝试抄，简单调整（不知道命名规制按猜想的来）
cat >> .config <<EOF
          #CONFIG_TARGET_x86=y
          #CONFIG_TARGET_x86_64=y
          #CONFIG_TARGET_x86_64_Generic=y
           CONFIG_TARGET_MediaTek_Ralink_MIPS=y
           CONFIG_TARGET_MT7621_based_boards=y
           CONFIG_TARGET_Newifi_D2=y
          EOF
编辑成功，但是还是x86的

KFERMercer, 2019-12-05:
> #CONFIG_TARGET_x86=y
> #CONFIG_TARGET_x86_64=y
> #CONFIG_TARGET_x86_64_Generic=y

这三行应当直接删掉.

closed 2019-12-05 by KFERMercer

fzsimlu, 2019-12-06:
去掉也不行

KFERMercer, 2019-12-06:
```
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2=y
```

azgh, 2019-12-25:
> ```
> CONFIG_TARGET_ramips=y
> CONFIG_TARGET_ramips_mt7621=y
> CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2=y
> ```

哪里可以找到全部的配置文件或者规则？如果要编译mir3g或者树莓派4b，该如何修改配置？