KFERMercer/OpenWrt-CI #18: 修改好同步的文件的分支字段，但是运行后没有同步最新源代码

bonnyandsky, 2019-12-09:
upstream已经编译成功了
[image]
然后我看了下分支ledenew下面的源代码还是和master一样的，并没有跟新coolsnowwolf的最新源码
[image]
这个是从你这里复制过来的2个脚本文件
[image]
自动同步的文件已经把相关的master替换成ledenew了
[image]
请问是哪里的问题？还是说cron：30  19 × × ×这里是表示要每天19点30分才会自动同步？

bonnyandsky, 2019-12-09:
我尝试着把utc时间改了下，然而他还是没有同步到最新的源码

[image]

KFERMercer, 2019-12-09:
> 请问是哪里的问题？还是说cron：30 19 × × ×这里是表示要每天19点30分才会自动同步？

utc时间. 换算成北京时间是凌晨三点半.

> 我尝试着把utc时间改了下，然而他还是没有同步到最新的源码

push函数的问题. 写脚本初期没有考虑到会有人放进非master分支. 有时间我会修复这个问题.

bonnyandsky, 2019-12-09:
> > 请问是哪里的问题？还是说cron：30 19 × × ×这里是表示要每天19点30分才会自动同步？
> 
> utc时间. 换算成北京时间是凌晨三点半.
> 
> > 我尝试着把utc时间改了下，然而他还是没有同步到最新的源码
> 
> push函数的问题. 写脚本初期没有考虑到会有人放进非master分支. 有时间我会修复这个问题.

我这也是被逼的，为了确保有一个分支能编译出正常使用的固件，才学着做一个新的分支用来同步新的源码吃螃蟹用。期待修复，感谢。

bonnyandsky, 2019-12-09:
对了，我去尝试下分支里编译固件的yml有没有问题。如果有，我再来反馈

KFERMercer, 2019-12-09:
已修复

closed 2019-12-09 by KFERMercer