KFERMercer/OpenWrt-CI #21: 建议将脚本定制部分提取出来

garypang13, 2019-12-10:
建议把脚本定制部分的内容提出来放到根目录某个文件中,好处:
    1.脚本文件更简洁清晰,  不再需要cat >> .config <<EOF之类的语法,根目录的配置文件直接按照.config的语法即可,只需要添加需要定制的部分,也不用拷贝整个.menuconfig 
    2.更方便的编译和管理多个设备的不同固件,只需要在根目录添加几个配置文件即可.

另外建议引入diy.sh,为高阶用户提供更多玩法, 比如https://github.com/garypang13/lede/raw/master/diy.sh

KFERMercer, 2019-12-10:
https://github.com/coolsnowwolf/lede/issues/2288#issuecomment-561700760

closed 2019-12-10 by KFERMercer

garypang13, 2019-12-10:
> [coolsnowwolf/lede#2288 (comment)](https://github.com/coolsnowwolf/lede/issues/2288#issuecomment-561700760)

根目录配置文件中只添加了定制的内容,仅仅只是把.yml中定制部分提取出来了而已,并不会造成任何后期依赖混乱

根目录配置文件不需要命名为.config,任何名字都可以,在.yml中重命名为.config即可,并不会随着克隆到本地而取代默认配置.

KFERMercer, 2019-12-11:
> 脚本文件更简洁清晰, 不再需要cat >> .config <<EOF之类的语法,根目录的配置文件直接按照.config的语法即可,只需要添加需要定制的部分,也不用拷贝整个.config文件

你不觉得这种做法是南辕北辙绕了一圈又回来了吗?

> 更方便的编译和管理多个设备的不同固件,只需要在根目录添加几个配置文件即可.

为什么不直接多部署几个flow?

> 另外建议引入diy.sh,为高阶用户提供更多玩法, 比如https://github.com/garypang13/lede/raw/master/diy.sh

耦合度太高. 高阶用户会直接修改源码.

> 根目录配置文件中只添加了定制的内容,仅仅只是把.yml中定制部分提取出来了而已,并不会造成任何后期依赖混乱

同第一条

> 根目录配置文件不需要命名为.config,任何名字都可以,在.yml中重命名为.config即可,并不会随着克隆到本地而取代默认配置.

然后你的库里就会多出一堆没用的文件

garypang13, 2019-12-14:
> > 脚本文件更简洁清晰, 不再需要cat >> .config <<EOF之类的语法,根目录的配置文件直接按照.config的语法即可,只需要添加需要定制的部分,也不用拷贝整个.config文件
> 
> 你不觉得这种做法是南辕北辙绕了一圈又回来了吗?
> 
> > 更方便的编译和管理多个设备的不同固件,只需要在根目录添加几个配置文件即可.
> 
> 为什么不直接多部署几个flow?
> 
> > 另外建议引入diy.sh,为高阶用户提供更多玩法, 比如https://github.com/garypang13/lede/raw/master/diy.sh
> 
> 耦合度太高. 高阶用户会直接修改源码.
> 
直接修改源码的话,如果lede的源码有更新,自己修改过的文件就没办法pull过来了,会造成很多不可控的问题.

peter-tank, 2019-12-18:
正如我做的：依官版，做了些hot patch源码和feeds的工作，[目前正尝试cache](https://github.com/peter-tank/openwrt/blob/v19.07.0-rc2_netgear_wndr/diy.sh)
问题：更新package/feeds的link，好像不管用？
我是打算替换shadowsocks-libev为bleeding，看起来链接修改成功了，但结果出来还是feeds/下的官版。

----
到底是需要cache哪些目录呢？
看编译后的目录结构，好像相当多关键目录超大，无法全部cache加速。