KFERMercer/OpenWrt-CI #32: 请问关于同步上游和编译脚本不会自动运行的问题

bonnyandsky, 2019-12-15:
之前问过,但是没说清楚.这边特地重新弄了下
下图是我昨天早上8点前最后一次修改Merge-upstream文件的截图
[image]
下图是我actions今天查看的截图
[image]
这个链接是我的Merge-upstream文件内容链接,时间我改成了UTC每天0点35执行，但是今天早上8点35也没有执行过
[https://github.com/bonnyandsky/lede/blob/ledenew/.github/workflows/Merge-upstream.yml](url)
还请k大帮忙看下是哪里出了问题，同样自动编译固件的yml也没有自动执行
[https://github.com/bonnyandsky/lede/blob/ledenew/.github/workflows/ledenew-ci.yml](url)

KFERMercer, 2019-12-15:
把00还成24试试

bonnyandsky, 2019-12-15:
> 把00改成24试试

但是ledenew-ci.yml的不是00也没运行啊……我是放在分支下面的，很奇怪

bonnyandsky, 2019-12-16:
今天看了下，还是没自动执行

KFERMercer, 2019-12-16:
🤔我认为是actions的bug

bonnyandsky, 2019-12-16:
大概真的是bug，我把同样的内容放在master分支下就可以自动运行了。只是修改了涉及到分支名字的地方

closed 2019-12-20 by KFERMercer