KFERMercer/OpenWrt-CI #36: 合并代码上游 workflows更新会导致更新失败？

kang-mk, 2019-12-20:
首先感谢大佬的同步和编译代码
我FORK过来的项目删除了L大自带的openwrt-ci.yml，然后新建了OpenWrt-X64-Actions.yml，编译一切顺利
之前同步上游代码运行一直正常，直到昨天L大更新了workflows目录的CI.yml文件
开始报错
2019-12-20T04:27:35.5001255Z shell: /bin/bash -e {0}
2019-12-20T04:27:35.5001376Z ##[endgroup]
2019-12-20T04:27:35.5197391Z ##[group]Run git pull https://github.com/coolsnowwolf/lede.git --no-edit --strategy-option ours
2019-12-20T04:27:35.5197612Z [36;1mgit pull https://github.com/coolsnowwolf/lede.git --no-edit --strategy-option ours[0m
2019-12-20T04:27:35.5223829Z shell: /bin/bash -e {0}
2019-12-20T04:27:35.5223934Z ##[endgroup]
2019-12-20T04:27:35.8634794Z From https://github.com/coolsnowwolf/lede
2019-12-20T04:27:35.8636500Z  * branch              HEAD       -> FETCH_HEAD
2019-12-20T04:27:36.1575731Z Auto-merging .github/workflows/OpenWrt-X64-Actions.yml
2019-12-20T04:27:36.3774251Z Merge made by the 'recursive' strategy.
2019-12-20T04:27:36.3828838Z  .github/workflows/OpenWrt-X64-Actions.yml          |  25 +-
2019-12-20T04:27:36.3829204Z  include/prereq-build.mk                            |   4 +-
2019-12-20T04:27:36.3829502Z  package/lean/luci-app-kodexplorer/Makefile         |   4 +-
2019-12-20T04:27:36.3829788Z  .../luasrc/controller/kodexplorer.lua              |  62 ++--
2019-12-20T04:27:36.3830498Z  .../luasrc/model/cbi/kodexplorer/api.lua           | 316 ++++++++++-----------
2019-12-20T04:27:36.3830898Z  .../luasrc/model/cbi/kodexplorer/settings.lua      |  52 ++--
2019-12-20T04:27:36.3831197Z  .../luasrc/view/kodexplorer/download.htm           |  27 +-
2019-12-20T04:27:36.3831306Z  .../luasrc/view/kodexplorer/version.htm            |  17 ++
2019-12-20T04:27:36.3831603Z  .../luci-app-kodexplorer/po/zh-cn/kodexplorer.po   | 102 +++++++
2019-12-20T04:27:36.3831895Z  .../root/etc/config/kodexplorer                    |  12 +-
2019-12-20T04:27:36.3832267Z  .../root/etc/init.d/kodexplorer                    |   2 +-
2019-12-20T04:27:36.3832767Z  package/lean/luci-theme-netgear/Makefile           |   2 +-
2019-12-20T04:27:36.3833042Z  .../htdocs/luci-static/netgear/css/style.css       |   2 +-
2019-12-20T04:27:36.3833579Z  13 files changed, 365 insertions(+), 262 deletions(-)
2019-12-20T04:27:36.3833892Z  create mode 100644 package/lean/luci-app-kodexplorer/luasrc/view/kodexplorer/version.htm
2019-12-20T04:27:36.4648703Z ##[group]Run git push origin $DOWNSTREAM_BRANCH
2019-12-20T04:27:36.4648902Z [36;1mgit push origin $DOWNSTREAM_BRANCH[0m
2019-12-20T04:27:36.4681102Z shell: /bin/bash -e {0}
2019-12-20T04:27:36.4681213Z env:
2019-12-20T04:27:36.4681292Z   DOWNSTREAM_BRANCH: master
2019-12-20T04:27:36.4681381Z ##[endgroup]
2019-12-20T04:27:39.0509189Z To https://github.com/kang-mk/lede
2019-12-20T04:27:39.0509704Z  ! [remote rejected]   master -> master (refusing to allow a bot to create or update workflow `.github/workflows/OpenWrt-X64-Actions.yml`)
2019-12-20T04:27:39.0510200Z error: failed to push some refs to 'https://github.com/kang-mk/lede'
2019-12-20T04:27:39.0529354Z ##[error]Process completed with exit code 1.
2019-12-20T04:27:39.0552519Z Cleaning up orphan processes
看错误是OpenWrt-X64-Actions.yml引起的

然后我删除了OpenWrt-X64-Actions.yml文件

重新激活同步
然后开始报
2019-12-20T04:36:22.0042205Z Updating files: 100% (10835/10835)
2019-12-20T04:36:22.0042312Z Updating files: 100% (10835/10835), done.
2019-12-20T04:36:22.0293295Z Reset branch 'master'
2019-12-20T04:36:22.0293648Z Branch 'master' set up to track remote branch 'master' from 'origin'.
2019-12-20T04:36:22.1117902Z [command]/usr/bin/git log -1
2019-12-20T04:36:22.1152663Z commit e14c703fc260afcc8a8befccc06349250951fe8b
2019-12-20T04:36:22.1153615Z Author: 尐ˊ;栤嚸┌ <20243226+kang-mk@users.noreply.github.com>
2019-12-20T04:36:22.1154100Z Date:   Fri Dec 20 12:35:52 2019 +0800
2019-12-20T04:36:22.1154181Z 
2019-12-20T04:36:22.1154850Z     Delete OpenWrt-X64-Actions.yml
2019-12-20T04:36:22.1166473Z Removed matchers: 'checkout-git'
2019-12-20T04:36:22.1404461Z ##[group]Run git config --global user.email "KFER.Mercer@gmail.com"
2019-12-20T04:36:22.1404810Z [36;1mgit config --global user.email "KFER.Mercer@gmail.com"[0m
2019-12-20T04:36:22.1404913Z [36;1mgit config --global user.name "KFERMercer"[0m
2019-12-20T04:36:22.1438569Z shell: /bin/bash -e {0}
2019-12-20T04:36:22.1438689Z ##[endgroup]
2019-12-20T04:36:22.1632392Z ##[group]Run git pull https://github.com/coolsnowwolf/lede.git --no-edit --strategy-option ours
2019-12-20T04:36:22.1632587Z [36;1mgit pull https://github.com/coolsnowwolf/lede.git --no-edit --strategy-option ours[0m
2019-12-20T04:36:22.1661652Z shell: /bin/bash -e {0}
2019-12-20T04:36:22.1661759Z ##[endgroup]
2019-12-20T04:36:22.5041904Z From https://github.com/coolsnowwolf/lede
2019-12-20T04:36:22.5042865Z  * branch              HEAD       -> FETCH_HEAD
2019-12-20T04:36:22.5958404Z CONFLICT (modify/delete): .github/workflows/openwrt-ci.yml deleted in HEAD and modified in 0f0c3f1a251f636df68050b75e296c6cc6965590. Version 0f0c3f1a251f636df68050b75e296c6cc6965590 of .github/workflows/openwrt-ci.yml left in tree.
2019-12-20T04:36:22.6288691Z Automatic merge failed; fix conflicts and then commit the result.
2019-12-20T04:36:22.6308982Z ##[error]Process completed with exit code 1.
2019-12-20T04:36:22.6331373Z Cleaning up orphan processes

好像是workflows目录文件名不统一引起的？（错误测试了几次之后直接删除了项目，重新fork了，忘了测试把名字改成和L大的一样会不会再次错误了）
或者能不能合并上游的时候直接排除这个目录

KFERMercer, 2019-12-20:
不要删除已有文件.

自动合并提交的ci无法得知哪些更改是需要保留的, 哪些是需要被覆盖的. 当出现合并冲突时, 会默认保留属于你的更改. 但是如果你删除文件, ci就无法操作.

其中涉及到git merge机制, 请自行了解.

自动合并ci能力有限, 不是, 也无法成为全能合并工具.

closed 2019-12-20 by KFERMercer

kang-mk, 2019-12-20:
ok,明白，那干脆就还在原有文件上进行修改了

kang-mk, 2019-12-20:
感谢大佬解答
是不是也就是说即便我保持openwrt-ci.yml文件不变，但是只要修改过，或者新增了Merge-Upstream.yml这个工作文件，也一样会导致这个错误？

adaaaaaa, 2019-12-21:
> 解决方案1: 自己在本地手动合并lean的修改.

是不是要改到和lede里面的workflow一样？？🤪🤪🤪
Merge-Upstream.yml要改吗？怎么弄？

adaaaaaa, 2019-12-21:
> "禁止套娃"

因为github限制，已经无法再自动更新了？