KFERMercer/OpenWrt-CI #39: 如何在编译中加入OpenClash或直接加入ipk与文件

HinataKato, 2020-01-07:
https://github.com/vernesong/OpenClash
需要先编译工具po2lmo，试过直接在Action的任务里面
git clone 然后EOF回环里面加luci-app-openclash，并不会编译出来

HinataKato, 2020-01-10:
已解决
加入
- name: 编译po2lmo
        run: |
          mkdir package/luci-app-openclash
          cd package/luci-app-openclash
          git init
          git remote add -f origin https://github.com/vernesong/OpenClash.git
          git config core.sparsecheckout true
          echo "luci-app-openclash" >> .git/info/sparse-checkout
          git pull origin master
          git branch --set-upstream-to=origin/master master
          pushd luci-app-openclash/tools/po2lmo
          make && sudo make install
          popd
      
后在下面的EOF回环里面加入
'CONFIG_PACKAGE_luci-app-openclash=n'

closed 2020-01-10 by HinataKato

KFERMercer, 2020-01-10:
你搞得好麻烦.
直接:
```
git clone --depth=1 https://github.com/vernesong/OpenClash.git ./package/luci-app-openclash
```

```
CONFIG_PACKAGE_luci-app-openclash=y
```
即可.

但是我并不鼓励这种在ci文件里拉取第三方插件的做法.

jif739519, 2020-06-20:
已解决
加入

```
name: 编译po2lmo
run: |
mkdir package/luci-app-openclash
cd package/luci-app-openclash
git init
git remote add -f origin https://github.com/vernesong/OpenClash.git
git config core.sparsecheckout true
echo "luci-app-openclash" >> .git/info/sparse-checkout
git pull origin master
git branch --set-upstream-to=origin/master master
pushd luci-app-openclash/tools/po2lmo
make && sudo make install
popd
后在下面的EOF回环里面加入
CONFIG_PACKAGE_luci-app-openclash=y
```
大佬，求帮助。
上面这些代码应该加在下面这个文件的什么位置？
https://github.com/KFERMercer/OpenWrt-CI/blob/master/openwrt-ci.yml