KFERMercer/OpenWrt-CI #41: 请问如何编译出带factory.bin的固件

bonnyandsky, 2020-01-10:
# 多文件系统支持:
          # cat >> .config <<EOF
          # CONFIG_PACKAGE_kmod-fs-nfs=y
          # CONFIG_PACKAGE_kmod-fs-nfs-common=y
          # CONFIG_PACKAGE_kmod-fs-nfs-v3=y
          # CONFIG_PACKAGE_kmod-fs-nfs-v4=y
          # CONFIG_PACKAGE_kmod-fs-ntfs=y
          # CONFIG_PACKAGE_kmod-fs-squashfs=y
          # EOF
这边我保持默认，但是编译出来的固件没有factory.bin的版本。请问是需要如何修改可以得到对应的固件。另外ipv6保持默认的话也是不开启状态么？
[image]

KFERMercer, 2020-01-10:
发下你的配置文件

bonnyandsky, 2020-01-13:
> 发下你的配置文件

#
# This is free software, lisence use MIT.
# 
# Copyright (C) 2019 P3TERX <https://p3terx.com>
# Copyright (C) 2019 KFERMercer <KFER.Mercer@gmail.com>
# 
# <https://github.com/KFERMercer/OpenWrt-CI>
# 

name: ledenew-ci

on:
  schedule:
    - cron: 30 18 * * *
  push:
    branches: 
      - master

jobs:
  build:
    runs-on: ubuntu-latest

    steps:
      - name: Checkout
        uses: actions/checkout@master
        with:
          ref: master

      - name: Initialization environment
        env:
          DEBIAN_FRONTEND: noninteractive
        run: |
          docker rmi `docker images -q`
          echo "Deleting files, please wait ..."
          sudo rm -rf /usr/share/dotnet /etc/apt/sources.list.d/* /etc/mysql /etc/php
          sudo -E apt-get -y purge azure-cli ghc* zulu* hhvm llvm* firefox google* dotnet* powershell openjdk* mysql* php*
          sudo -E apt-get update
          sudo -E apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler
          sudo -E apt-get -y autoremove --purge
          sudo -E apt-get clean
      - name: Update feeds
        run: |
          ./scripts/feeds update -a
          ./scripts/feeds install -a
      - name: Costom configure file
        run: |
          rm -f ./.config*
          touch ./.config
          #
          # ========================固件定制部分========================
          # 
          # 
          # 如果不对本区块做出任何编辑, 则生成默认配置固件. 
          # 
          # 以下为定制化固件选项和说明:
          #
          #
          # 有些插件/选项是默认开启的, 如果想要关闭, 请参照以下示例进行编写:
          # 
          #          =========================================
          #         |  # 取消编译VMware镜像:                   |
          #         |  cat >> .config <<EOF                   |
          #         |  # CONFIG_VMDK_IMAGES is not set        |
          #         |  EOF                                    |
          #          =========================================
          #
          # 
          # 以下是一些提前准备好的一些插件选项.
[... 235 lines cut ...]
CONFIG_PACKAGE_strongswan-mod-sha1=y
CONFIG_PACKAGE_strongswan-mod-socket-default=y
CONFIG_PACKAGE_strongswan-mod-stroke=y
CONFIG_PACKAGE_strongswan-mod-updown=y
CONFIG_PACKAGE_strongswan-mod-x509=y
CONFIG_PACKAGE_strongswan-mod-xauth-generic=y
CONFIG_PACKAGE_strongswan-mod-xcbc=y
CONFIG_PACKAGE_trojan=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_v2ray=y
CONFIG_PACKAGE_vnstat=y
CONFIG_PACKAGE_vnstati=y
# CONFIG_PACKAGE_vsftpd-alt is not set
CONFIG_PACKAGE_watchcat=y
CONFIG_STRONGSWAN_ROUTING_TABLE="220"
CONFIG_STRONGSWAN_ROUTING_TABLE_PRIO="220"
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
以下是相关设置
[image]
[image]
[image]
[image]

closed 2020-05-06 by KFERMercer