KFERMercer/OpenWrt-CI #55: 最近云编译一直出错

NeedforGit, 2020-04-11:
最近云编译一直出错，用的示例里面的 yml 文件 （https://github.com/KFERMercer/OpenWrt-by-lean/blob/CI-demo/.github/workflows/openwrt-ci.yml） 已经改过 master 了，之前一直没问题，就是最近不行，更新Lean大的源码后，一直编译报错。

求解决！

miroda, 2020-04-18:
最近这个 脚本不行了

imymi, 2020-04-24:
我的提示如下

`make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [world] Error 2
/home/runner/work/lede/lede/include/toplevel.mk:225: recipe for target 'world' failed
##[error]Process completed with exit code 2.`

herowhere, 2020-04-28:
我也是这三天一直出错，怎么处理？

KFERMercer, 2020-04-28:
今天修

andychinlab, 2020-05-06:
> 今天修

大神，本人小白，最近刚研究在线编译，发现L大的workflow里代码和您的CI代码不一样，全部拷贝您的代码到L大workflow里的CI会编译报错，而且L大的workflow里没有自定义功能部分，现在完全一头雾水，求解答，感激

KFERMercer, 2020-05-06:
已修. 顺便重申一下, 编译op建议去使用 [P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)

closed 2020-05-06 by KFERMercer