KFERMercer/OpenWrt-CI #56: 请问下如何同步合并上游的多个分支

kang-mk, 2020-04-20:
比如上游代码里面有多个分支的，默认只能同步默认的分支，而其他分支不能同步，修改了

    - name: 效验本地文件
      uses: actions/checkout@master
      with:
        ref: 分支1名字
        fetch-depth: 0
        lfs: true

    - name: 效验本地文件
      uses: actions/checkout@master
      with:
        ref: 分支2名字
        fetch-depth: 0
        lfs: true

这两个地方，第一步前面两个对比都可以顺利执行成功
    - name: 加载上游提交
      run: git pull https://github.com/Lienol/openwrt-luci.git --no-edit --strategy-option ours
然后到这一步就失败了,不知道这里应该怎么修改了
然后下面接着的

    - name: 推送上游提交
      env:
        DOWNSTREAM_BRANCH: 分支1名字
      run: git push origin $DOWNSTREAM_BRANCH

    - name: 推送上游提交
      env:
        DOWNSTREAM_BRANCH: 分支2名字
      run: git push origin $DOWNSTREAM_BRANCH

自然也就直接跳过不执行了

求帮助，非常谢谢

KFERMercer, 2020-05-06:
checkout 然后分别拉取上游, 最后分别提交. 同时加载多个分支会引发错误.

closed 2020-05-06 by KFERMercer

kang-mk, 2020-05-07:
搞定了，拉取上游的地方加一个分支参数就好了  谢谢