KFERMercer/OpenWrt-CI #59: 请问，可以将项目更新至自己项目的文件夹吗

authon, 2020-05-15:
请问，可以将项目更新至自己项目的文件夹吗

KFERMercer, 2020-05-15:
啥意思

authon, 2020-05-16:
> 啥意思

就是，例如luci-app-ssr这个项目，放到我的package项目里面，package里面有很多插件，然后让他自动更新

KFERMercer, 2020-05-17:
可以修改这个: https://github.com/KFERMercer/OpenWrt/blob/master/.github/workflows/update-helloworld.yml

closed 2020-05-17 by KFERMercer

authon, 2020-05-19:
> 可以修改这个: https://github.com/KFERMercer/OpenWrt/blob/master/.github/workflows/update-helloworld.yml

另外一个又出现这个
Post job cleanup.
/usr/bin/git version
git version 2.26.2
/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
fatal: No url found for submodule path 'luci-app-serverchan/luci-app-serverchan' in .gitmodules
##[warning]The process '/usr/bin/git' failed with exit code 128

这是代码
    - name: Checkout
      uses: actions/checkout@master
      with:
        ref: master
        fetch-depth: 0
        lfs: true

    - name: Set git identity
      run : |
        git config --global user.email "KFER.Mercer@gmail.com"
        git config --global user.name "KFERMercer"
    - name: Load latest luci-app-serverchan
      run: |
        git clone https://github.com/tty228/luci-app-serverchan.git --depth=1 ./tmp/luci-app-serverchan
        cp -rf ./tmp/luci-app-serverchan ./luci-app-serverchan
    - name: Apply commit changes
      run: |
        git add ./luci-app-serverchan/
        echo -e "[bot] luci-app-serverchan: update\n\nlatest commit: $(cat ./tmp/luci-app-serverchan/.git/refs/heads/master)" > ./tmp/message
        git commit --file="./tmp/message" --author="github-actions[bot] <41898282+github-actions[bot]@users.noreply.github.com>" || exit 0
    - name: Push Commits
      run: git push origin master

ssr的按照你的改了目录就行，这个改了目录后不行