KFERMercer/OpenWrt-CI #67: 定制了插件，但编译后的固件没有SSR插件？

weonbean, 2020-08-06:
在插件选择中选了所有的插件，但定制出来的固件中，却没有SSR插件，其他插件都有。请问是换了名字还是什么？
          # 常用LuCI插件选择:
          cat >> .config <<EOF
          CONFIG_PACKAGE_luci-app-adbyby-plus=y
          CONFIG_PACKAGE_luci-app-aria2=y
          CONFIG_PACKAGE_luci-app-baidupcs-web=y
          CONFIG_PACKAGE_luci-app-docker=y
          CONFIG_PACKAGE_luci-app-frpc=y
          CONFIG_PACKAGE_luci-app-kodexplorer=y
          CONFIG_PACKAGE_luci-app-minidlna=y
          CONFIG_PACKAGE_luci-app-openvpn=y
          CONFIG_PACKAGE_luci-app-openvpn-server=y
          CONFIG_PACKAGE_luci-app-qbittorrent=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Socks=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y
          CONFIG_PACKAGE_luci-app-ttyd=y
          CONFIG_PACKAGE_luci-app-v2ray-server=y
          CONFIG_PACKAGE_luci-app-verysync=y
          CONFIG_PACKAGE_luci-app-webadmin=y
          CONFIG_PACKAGE_luci-app-wireguard=y
          CONFIG_PACKAGE_luci-app-wrtbwmon=y
          EOF

另外，出现了如下的warning，但最后还是能生成artifacts，

 Check warning on line 1 in .github

@github-actions
github-actions
/ Build OpenWrt

.github#L1
No files were found for the provided path: ./bin/targets/*/*/config.seed. No artifacts will be uploaded.

adwgjad, 2020-08-23:
CONFIG_PACKAGE_luci-app-ssr-plus=y

adwgjad, 2020-08-23:
你只定制了ssr-plus的依赖，你的ssr-plus依赖不定制也可以，直接定制CONFIG_PACKAGE_luci-app-ssr-plus=y就可以自动把依赖编译进去。简洁一点，不用看的眼花

cnhap, 2020-08-25:
> 你只定制了ssr-plus的依赖，你的ssr-plus依赖不定制也可以，直接定制CONFIG_PACKAGE_luci-app-ssr-plus=y就可以自动把依赖编译进去。简洁一点，不用看的眼花

加上CONFIG_PACKAGE_luci-app-ssr-plus=y 编译出来 也没有

https://github.com/cnhap/lede/blob/master/.github/workflows/myopenwrt.yml

weonbean, 2020-09-03:
> 你只定制了ssr-plus的依赖，你的ssr-plus依赖不定制也可以，直接定制CONFIG_PACKAGE_luci-app-ssr-plus=y就可以自动把依赖编译进去。简洁一点，不用看的眼花

非常感谢！只定制了依赖没定制本体，傻了。。

weonbean, 2020-09-05:
> > 你只定制了ssr-plus的依赖，你的ssr-plus依赖不定制也可以，直接定制CONFIG_PACKAGE_luci-app-ssr-plus=y就可以自动把依赖编译进去。简洁一点，不用看的眼花
> 
> 加上CONFIG_PACKAGE_luci-app-ssr-plus=y 编译出来 也没有
> 
> https://github.com/cnhap/lede/blob/master/.github/workflows/myopenwrt.yml

在diy-part1.sh中，要去掉示例前面的#，大雕把插件隐藏了
# Uncomment a feed source
sed -i 's/^#\(.*helloworld\)/\1/' feeds.conf.default

cnhap, 2020-09-05:
> > > 你只定制了ssr-plus的依赖，你的ssr-plus依赖不定制也可以，直接定制CONFIG_PACKAGE_luci-app-ssr-plus = y就可以自动把依赖编译进去。简洁一点，不用看的眼花
> > 
> > 
> > 加上CONFIG_PACKAGE_luci-app-ssr-plus = y编译出来也没有
> > https://github.com/cnhap/lede/blob/master/.github/workflows/myopenwrt.yml
> 
> 在diy-part1.sh中，要去掉示例前面的＃，大雕把插件隐藏了
> 
> # 取消注释供稿源
> sed -i's /^#(.* helloworld）/ \ 1 /'feeds.conf.default

没有看明白

Sgrass29, 2020-10-21:
- name: Update feeds
        run: |
          sed -i 's/#src-git helloworld/src-git helloworld/g' ./feeds.conf.default
          ./scripts/feeds update -a
          ./scripts/feeds install -a

在那里加          sed -i 's/#src-git helloworld/src-git helloworld/g' ./feeds.conf.default